CougsInSpace/CougSat1-Software #11: IHU System Interface ⇄ PMIC

WattsUp, 2018-04-12:


kevinhikaruevans, 2018-09-09:
I'm not on the Github organization yet, but I'll be on this one too. 👍

kevinhikaruevans, 2019-01-20:
Finally starting to tackle this... I'm still a little unsure if I'm doing this right, so can somebody check over if this is the correct steps for this task?

1. For each command listed in the [PMIC documentation](https://github.com/CougsInSpace/CougSat1-Software/blob/master/CougSat1-PMIC/docs/PMICInterface.pdf), create a function in IHU/src/systemInterfaces/PMIC.cpp
2. Each function accepts an argument
3. Convert the arguments to match the command formats
4. Send the data over i2c to the PMIC address

kevinhikaruevans, 2019-09-22:
I am reopening this, as the current master does not have any changes from PR https://github.com/CougsInSpace/CougSat1-Software/pull/283.

PR #283 (2019-05-20): IHU <-> PMIC interface for I2C requests
Fixes #11

There might also be a [typo in the docs](https://github.com/CougsInSpace/CougSat1-Software/blob/master/CougSat1-PMIC/docs/PMICInterface.pdf) on 2.4: Temperature data request: the command identifier is 0x03, but that is already used for current data requests. Presumably, it should be 0x04?