CougsInSpace/CougSat1-Software #20: Design ADCS Control Algorithm

WattsUp, 2018-04-12:


phillip-whitworth, 2018-08-20:
This would be the work flow of the main loop.  A graphic flow chart of the system might be enough with some key points noted on a word document.

phillip-whitworth, 2018-08-27:
[image]

phillip-whitworth, 2018-09-18:
[image]
1.	The log is the storage of data that can be access for later use.  This can be volatile memory.
2.	The IHU is controlled by another system and is only listed here as the interface between the two systems.  
3.	The sensor data is retrieved from the 4 IHUs and GPS as well as other systems that can offer some sensing capabilities.  These systems could be the voltages from the solar panels or some thermal sensor data from around the exterior of the satellite.
4.	The error checking routine is the internal watchdog for the ADCS.  This system will collect data from the system and possibly apply some routines to assess the system causing the error.  It will also be the main location for storing the error list for the ADCS.
5.	The Main Loop is a Que system that manages the ADCS based on priorities of the satellite operations.  
6.	The attitude calculations function will receive all the sensor data and compute the satellite’s attitude in both the earth and body frames.  
7.	The control loop will receive the current attitude and then devise an error based on the target attitude to the current attitude.  These errors will be in body frame and then translated to the three magnetorquer coils via a PID control process.
8.	The Orbital projection function will take in the attitude of the satellite and predict the path and expected conditions of the next few orbits.  This will allow for some timing to be achieved on when the satellite is over a location on earth or needing to point at a given target.   
9.	The data block is a mostly non-volatile storage block that contains the lookup tables for gravity field measurements and celestial body locations to aid in determining the satellite’s attitude. 

[Main protocol for attitude control.pdf](https://github.com/CougsInSpace/CougSat1-Software/files/2394632/Main.protocol.for.attitude.control.pdf)

closed 2018-09-18 by phillip-whitworth