CougsInSpace/CougSat1-Software #108: IHU: Design Event pmicMessage

WattsUp, 2018-07-02:


kevinhikaruevans, 2019-09-24:
@WattsUp 

Just for clarity: is this an event that gets pushed into the event queue when a PMIC sends data to the IHU over i2c? What would be the best way to get specifics on the design of it?

WattsUp, 2019-09-28:
@kevinhikaruevans Yes... maybe...
The IHU talks to the PMIC when grabbing telemetry or commanding an operation to turn on/off a rail. There is an interrupt request pin driven by any of the slaves to alert the IHU that they would like to talk. That interrupt would add an event to check each subsystem and ask what is up my dude (the hardware doesn't know which one want to talk). I guess there isn't a need for pmicMessage to be an event, the IHU could talk to it in either collectTelemetry or subsystemIRQCheck.

That would mean the only events the IHU needs are:
- collectTelemetry
- subsystemIRQCheck
- ihuPeriodic - performs a health check, asks each subsystem if they are okay, checks C&DH hardware, etc., adjust germination payload if going (heaters, lighting)
- sendSegment - to comms for transmitting, moving layer 4 from comms to layer 5 in C&DH [CougSatNet](https://github.com/CougsInSpace/CougSat1-Software/blob/master/CougSat1-Comms/docs/CougSatNet.pdf)
- getSegment - same as send but reversed

And all the command events which could occur at a set datetime in the future, probably maintain a timestamped queue in the file system and check that every so often
- payloadCameraExecute - take pictures
- payloadGerminationDoStuff - release the water
- payloadPeriodic - log radiation counts, temperatures, germination sensors, etc.
- adcsAdjust - point at Pullman, Area 51, etc.

kevinhikaruevans, 2019-09-29:
Sorry for drawing this one out a ton, but there's a lot that I feel like I don't know. Does this seem like an OK plan for this task and others like it? 

- On the IHU, have an active high `InterruptIn` on `BUS_I2C1_IRQ` and `BUS_I2C0_IRQ` to trigger a handler. The handler would call `subsystemIRQCheck` (presumably by enqueuing it on the main thread?)
- In `subsystemIRQCheck`, iterate through each subsystem and request the status of each.
   - Question: Is there a status request defined? How does the IHU know whether or not to do something? Are these prioritized as well?

WattsUp, 2019-09-30:
Hmm. Maybe we need a whiteboarding session and figure it out, probably through in Micah too. Does after the Wednesday meeting work?

Bullet point 1 sounds correct.
Bullet point 2: I don't think there is a status request defined. Only the [ADCS](https://photos.app.goo.gl/L25mPFSgqSsRjzCw8) and [PMIC](https://github.com/CougsInSpace/CougSat1-Software/blob/master/CougSat1-PMIC/docs/PMICInterface.pdf) have been written but not implemented. I suspect that they need to be updated (redesigned??) to reflect the current hardware and/or subsystem design. There could also be a generic register mapping for each subsystem that the interface extends (like register 00 is status, register 01 is temps, etc) IDK how much of that will be common between subsystems so it might not be worth it.