CougsInSpace/CougSat1-Software #257: ADCS: Research H-Bridge datasheet

phillip-whitworth, 2018-11-05:
this leads into """ADCS: Design Magnetorquer Class #17"""

Determine the pins used, frequency of transmissions, type of signal needed to control the current flow.

phillip-whitworth, 2019-02-05:
this is completed.  not sure what we should do to close it.  We are starting on the design now

WattsUp, 2019-02-06:
Just answer the questions asked in the issue and close it.

jlung21, 2019-02-09:
Pins used for logic: 
Pin 5 - IN1 
Pin 6 - IN2 
Pin 7 -nSleep

Other pins:

Pin 1 - Vm (Motor power supply)
Pin 2 - Output 1
Pin 3 - Output 2
Pin 4 - GND
Pin 8 - Vcc (Logic power supply)

Externally Applied PWM Frequency Conditions 0 - 250 kHz (Min - Max) According to manufacturer

Logic inputs take a range of 0 - 5.5 V (Min - Max) According to manufacturer

Current can be controlled by applying the PWM frequency range listed above to the IN1, and IN2 logic inputs. nSleep functions as the enable signal.

closed 2019-02-09 by WattsUp