CougsInSpace/CougSat1-Software #296: AVIONICS: Test H-Bridge Driver

WattsUp, 2019-04-11:


kevinhikaruevans, 2019-09-30:
Do we have any DRV8837s on-hand and mounted a breakout? If so, where could somebody pick one up to borrow for testing?

WattsUp, 2019-10-02:
There is a board for it in the electrical systems room. I can bring it tomorrow. Green LED is forward. Red LED is reverse. Red wire is 3.3V. Black is ground. Orange is nSleep. White is IN1. Grey is IN2.

kevinhikaruevans, 2019-10-06:
How do we want to handle non-blocking calls?

kevinhikaruevans, 2019-10-06:
@WattsUp nSleep is tied high on the board. Is that supposed to be constant high or is there a solder bridge somewhere? I expected it to be driven by the microcontroller.

WattsUp, 2019-10-07:
Since you are just writing PWM outputs that are pretty short, there wouldn't be a difference between blocking and non-blocking. Setting up something to handle writing the outputs would probably take longer than actually setting them.

I thought I could toggle it. It is driven by the microcontroller