CougsInSpace/CougSat1-Software #326: IHU: Implement filestorage system on SD Cards

micah-priddis, 2019-10-06:


christianrouhana, 2019-10-06:
Wiring complete. FAT32 file system created. Working on accessing specific files.

christianrouhana, 2019-10-13:
Writing to LittleFileSystem with HeapBlockDevice complete. Now working on LittleFileSystem with an external BlockDevice.

christianrouhana, 2019-10-14:
FATFileSystem and LittleFileSystem successfully implemented on SDBlockDevice with Nucleo-L476RG specifically. Successfully managed to format, create file, read the file, and then read the created file on restart.

WHOmagoo, 2019-10-17:
Can you please push your code? I'm looking to doing something similar and would like to see how you have done this.

christianrouhana, 2019-10-18:
Pushed to FILESYS branch.

CougSat1-Software/CougSat1-IHU/test/FATFileSystem/